polishx3/Craftbot-Prusaslicer #1: Print off the build plate - so it says

Brazedowl, 2023-01-13:
First off, this is AMAZING. I hate Craftware's slicer so very very much.

I imported and sliced up a pretty simple model in two colors, but when I bring the gcode over to the Idex it is VERY insistent that the model is off the build area. X: 18.00mm or some such at line 72.

Attaching model, PrusaSlicer Project,  and gcode. Don't laugh. I'm a teacher and making hallpasses for every teacher color-coded by section of the school. 

[PrusaProject.zip](https://github.com/polishx3/Craftbot-Prusaslicer/files/10416037/PrusaProject.zip)

Randy1727, 2023-02-19:
I have the same issue including M220 and M572 errors

polishx3, 2023-02-19:
@Brazedowl 

Your build area is too big; will have to edit build area to dual. The build area is for single

polishx3, 2023-02-20:
I have update build plate size

closed 2023-02-20 by polishx3