polishx3/Craftbot-Prusaslicer #3: M572 error,

Fisheye3D, 2023-06-13:
Love the profiles, When I have the Wipe tower activated I'm getting a M572 error in gcode, With it off its fine, Not sure why the wipe tower would have PA settings, Would adding some kind of gcode substitutions to replace M572 DO SO to nothing work?

[image]
 
[image]

polishx3, 2023-06-14:
Added with the latest updates to PrusaSlicer.
Unfortunately no PA on the craftbots; maybe later
Testing now to make sure substitutions don't break somethings else
should have update in couple days
thanks

polishx3, 2023-06-15:
Uploaded 0.04, repaired
thanks

closed 2023-06-15 by polishx3