polishx3/Craftbot-Prusaslicer #4: Bed size/offset incorrect?

Fisheye3D, 2023-06-19:
[image]
[image]
I think the bed size is incorrect, this is for left extruder only, shows in bed that its set to 425 with no offset,
[image]

polishx3, 2023-06-19:
Interesting craftware must have adjusted the print size in the new firmware, smaller print area.
Cause when I slice a large model in Craftware; their slicer. I get the same error.
Offset causes negative locations; which Craftbot Printer does not like.

polishx3, 2023-06-19:
Om second thought it appears that 400 is max; will adjust

Fisheye3D, 2023-06-19:
For this print i just changed it to the 386 listed on site. 
[image]

polishx3, 2023-06-19:
what a mess; updated the sizes
thanks

polishx3, 2023-06-19:
updated

closed 2023-06-19 by polishx3